labscript-suite/labscript-devices #59: Break circular dependencies in the labscript-suite

philipstarkey, 2020-06-25:
We currently have circular dependencies where labscript-devices depends on BLACS, and vice versa.

I think we can break this by migrating the contents of `__init__.py` over to labscript-utils. If this was then imported into labscript-devices it would also remain backwards compatible. BLACS (and runviewer) would then access the labscript-devices import machinery via labscript-utils, breaking the circular dependency.

philipstarkey, 2020-06-25:
This would also mean that installing runviewer didn't require installing blacs (runviewer depends on labscript-devices which depends on blacs)

dihm, 2021-11-30:
Everything is finally merged, so going to close this.

closed 2021-11-30 by dihm