labscript-suite/labscript-devices #79: PulseblasterESRPRO500 no runviewer_parser

AllenNCC, 2021-06-28:
I think the PulseBlasterESRPRO500.py under labsrcipt_devices forgot to implement runviewer_parser. 

I simply compare it with the PulseBlasterESRPRO200.py file and add them on my own.

matth2peters, 2021-06-29:
I also have an issue where I need to manually increase the number of flags to 24 so that I can set flags 21 and 23 high for a 10 ns pulse as described in pg 27 of the manual: http://www.spincore.com/CD/PulseBlasterESR/SP4/PBESR-Pro_Manual.pdf

If at least one of flags 21-23 are not set high, then the PB has a pulse length of 0 (no output), so I made digital outputs for flags 21-23 and set them to high. This fixes the problem, but then the runviewer quickfix allen listed above no longer works.

dihm, 2021-06-29:
> I also have an issue where I need to manually increase the number of flags to 24 so that I can set flags 21 and 23 high for a 10 ns pulse as described in pg 27 of the manual: http://www.spincore.com/CD/PulseBlasterESR/SP4/PBESR-Pro_Manual.pdf

I was actually about to ask this here, since this has always been my experience with any of the ESR Pro boards. I've been considering making a PR that makes the necessary change to expose those pins, but wanted to figure out if somebody somewhere had an ESR Pro board that actually works with the existing code (a distinct possibility since spincore does allow default behaviors to change between firmwares without warning). Knowing the existing code works for somebody will influence how the more permanent fix should work. @AllenNCC Do you also need to increase the number of flags, then set 21-23 high in order to see any output from any of the pins?

The corresponding runviewer fix that has always worked for me is to increase the number of flags for the runviewer as well. The total tweak I use in my lab is:
```python
class PulseBlasterESRPro500(PulseBlaster_No_DDS):
    description = 'SpinCore PulseBlaster ESR-PRO-500'
    clock_limit = 50.0e6 # can probably go faster
    clock_resolution = 4e-9
    n_flags = 24
    core_clock_freq = 500.0

@BLACS_tab    
class pulseblasteresrpro500(Pulseblaster_No_DDS_Tab):
    # Capabilities
    num_DO = 24
    def __init__(self,*args,**kwargs):
        self.device_worker_class = PulseblasterESRPro500Worker 
        Pulseblaster_No_DDS_Tab.__init__(self,*args,**kwargs)
    
    
class PulseblasterESRPro500Worker(PulseblasterNoDDSWorker):
    core_clock_freq = 500.0

    
@runviewer_parser
class PulseblasterESRPro500Parser(PulseBlaster_No_DDS_Parser):
    num_dds = 0
    num_flags = 24
```

AllenNCC, 2021-06-29:
Hi @dihm , 

I have a > 10 years old pulseblaster board (not PBESRPRO500). I recently sent it back for a firmware upgrade and according to the company, the firmware used is most similar to PBESRPRO500 without the short pulse feature. So I'm just using the PBESRPRO500 template but I don't need to deal with the issue mentioned here.

matth2peters, 2021-06-29:
> ```python
> class PulseBlasterESRPro500(PulseBlaster_No_DDS):
>     description = 'SpinCore PulseBlaster ESR-PRO-500'
>     clock_limit = 50.0e6 # can probably go faster
>     clock_resolution = 4e-9 [...]
I still have an issue with this code. When I try to use runviewer, the only channel that appears in the "channels" window is the clockline into my NI PCI 6713 (no analog outputs/digital outputs/etc.) It works fine when I keep everything the same and use a pineblaster clock as the clock.

I just started using labscript a couple weeks ago and haven't been able to figure out why this is, do you have a suggestion for where in the code I can look to change something?

Thanks!

philipstarkey, 2021-06-30:
@dihm I believe I originally developed the ESR labscript suite device code and tested on a ESR board loaned to us from another local university. So there are definitely devices out there with firmware that don't need those flags set. Whether or not people are actively using them with labscript, I have no idea though!

chrisjbillington, 2026-10-18:
Fixed by #153, merged 2026-10-18.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-10-18 by chrisjbillington