labscript-suite/labscript-devices #100: PyCapture2 error

philipstarkey, 2022-05-27:
I have a new problem,I am trying to solve it.
Here is my traceback
```
Exception in worker - Thu May 26, 22:05:37 :
Traceback (most recent call last):
  File "c:\labscript\lib\site-packages\labscript_devices\IMAQdxCamera\blacs_workers.py", line 251, in init
    self.camera = self.get_camera()
  File "c:\labscript\lib\site-packages\labscript_devices\IMAQdxCamera\blacs_workers.py", line 280, in get_camera
    return self.interface_class(self.serial_number)
  File "c:\labscript\lib\site-packages\labscript_devices\FlyCapture2Camera\blacs_workers.py", line 80, in __init__
    self.camera.connect(bus.getCameraFromSerialNumber(serial_number))
  File "src\cython\PyCapture2.pyx", line 3406, in PyCapture2.BusManager.getCameraFromSerialNumber
  File "src\cython\PyCapture2.pyx", line 3849, in PyCapture2.checkError
PyCapture2.Fc2error: b'Bus Master Failure.'

Fatal exception in main process - Thu May 26, 22:05:37 :
 Traceback (most recent call last):
  File "c:\labscript\lib\site-packages\blacs\tab_base_classes.py", line 837, in mainloop
    next_yield = inmain(generator.send,results)
  File "c:\labscript\lib\site-packages\qtutils\invoke_in_main.py", line 88, in inmain
    return get_inmain_result(_in_main_later(fn, False, *args, **kwargs))
  File "c:\labscript\lib\site-packages\qtutils\invoke_in_main.py", line 150, in get_inmain_result
    raise value.with_traceback(traceback)
  File "c:\labscript\lib\site-packages\qtutils\invoke_in_main.py", line 46, in event
    result = event.fn(*event.args, **event.kwargs)
  File "c:\labscript\lib\site-packages\blacs\tab_base_classes.py", line 536, in _initialise_worker
    raise Exception('Device failed to initialise')
Exception: Device failed to initialise
```
Thanks again for your help

_Originally posted by @Moriaty-TT in https://github.com/labscript-suite/labscript-suite/issues/68#issuecomment-1139422721_

dihm, 2022-05-28:
@Moriaty-TT Apologies for this not occurring to me sooner, but is your camera supported by the newer Spinnaker SDK (PySpin python wrapper)? When Point Grey was acquired by FLIR, they deprecated the old FlyCapture2 SDK in favor of Spinnaker, which is why you are stuck with Python 3.6. If you camera is supported by the newer SDK, I would highly recommend you switch so you can move to a newer version of python as well.

If it isn't, I can try to track down this particular issue, though no promises on the timeline. The best self-help I can recommend is picking through the C documentation of the SDK. It should provide a better explanation of the bus manager error. My quick guess is that it can't find your camera based on its serial number. You can test that theory by manually running the pycapture2 wrapper code from the blacs_device (in the terminal or a jupyter notebook) to try connecting to the camera outside of labscript. There should be a command for the Bus Manager to list all connected cameras which may help illuminate what is going on.

Moriaty-TT, 2022-06-01:
Thanks for your help.But I've given up to use pychapture2 for the time being,and I'm trying another plan based on labscript to realize my experiment.

chrisjbillington, 2022-09-25:
For what it's worth, I was working with a camera using PyCapture2 and Python 3.6 recently, and saw the "Bus Master Failure" error. The problem for us was that we had specified the serial number as a string, whereas the labscript device class expects an integer - but no code checks this in order to raise a TypeError.