labscript-suite/labscript-devices #107: PrawnBlaster Potential Bug with Waits

pmiller2022, 2023-08-04:
I have been setting up a digital output addition to the prawnblaster and when I add waits, the run on blacs lasts as long as the timeout(s) for the waits, even with an external trigger and the experiment ending well before the length of the timeout(s). Would there be any way for labscript to stop the prawnblaster early if the experiment ends?

philipstarkey, 2023-08-27:
Probably need some more information to debug this. Is it with a PrawnBlaster as the primary pseudoclock? Do you know which part of the code base is hanging for that timeout length?

dihm, 2023-08-27:
The prawn blaster is the primary (and only) pseudo clock. It was fairly easy to reproduce; set a PB with whatever intermediate device and minimal instructions and a wait with a timeout of 4 s. BLACS always waits the whole 4s before reporting the shot complete.

It's not totally clear where the hang is. If I remember correctly, the shot timing plugin reports the shot length to be the longer wait length from the start and counts it down accordingly, probably indicating something more BLACS related now that I think about it. I can check this behavior on a pulseblaster this week to help isolate a little better.